usnistgov/MIST #3: Crash on stiching

absurdvision, 2017-12-04:
The plugin crashes while the stiching/preview is done.

On running preview, it gets stuch at 89% and prints out in the error console

`Exception in thread "Thread-5" java.lang.ArrayIndexOutOfBoundsException: 8312651
	at ij.process.ColorProcessor.set(ColorProcessor.java:313)
	at gov.nist.isg.mist.lib.export.blend.OverlayBlend.blend(OverlayBlend.java:89)
	at gov.nist.isg.mist.lib.export.LargeImageExporter.exportImageNoOverlap(LargeImageExporter.java:268)
	at gov.nist.isg.mist.lib.executor.StitchingExecutor.previewNoOverlap(StitchingExecutor.java:1034)
	at gov.nist.isg.mist.lib.executor.StitchingExecutor.run(StitchingExecutor.java:198)
	at java.lang.Thread.run(Thread.java:745)
`
while log is at 
`Warning: Unable to load CUDA. Disabling CUDA execution option.
Checking args for preview:
Arg check passed
Preparing preview image size: 2883x2883`

On running stiching, it crashes again and program exits. The log file is attached.
[hs_err_pid20256.log](https://github.com/usnistgov/MIST/files/1526753/hs_err_pid20256.log)

The task is to stich a small 3x3 grid of jpeg images

mmajurski, 2017-12-05:
Are you willing to share your 3x3 grid of images along with the stitching params file?

I would like to reproduce the error to figure out what is causing the crash.

Thanks,
Michael

absurdvision, 2017-12-08:
I have uploaded the files in this comment.

[3x3_error.zip](https://github.com/usnistgov/MIST/files/1542430/3x3_error.zip)

mmajurski, 2017-12-08:
Your image grid appears to have zero overlap between the images tiles.
Filename pattern: {ccc}_{rrr}.jpeg

The error occurred because the image tiles are not all the same dimensions (width, height). They vary between 961 and 962. MIST only supports images of identical size per grid. 

I updated the wiki FAQ to include this information. 

Once I cropped your color images to all be 961x961x3 I was able to preview them using MIST. From my visual inspection, it appears that the stitched preview image (with zero overlap) is correct.

[image]

closed 2017-12-08 by mmajurski