usnistgov/MIST #4: Cannot use global position file

alexandrebastien, 2018-02-27:
Hi,

I have this error when I try to assemble my secondary channel using the metadata:
   Unable to parse correlation for C1-000.tif
   Error parsing absolute positions: C:\Users\ALBAS18\Desktop\stitch\C1-global-positions-0.txt
   Error unable to find file: Error parsing: C:\Users\ALBAS18\Desktop\stitch\C1-global-positions-0.txt. 
   Skipping timeslice: 0
   Done

I have 3x 792 files in the same folder named C1-{ppp}.tif, C2-{ppp}.tif and C3-{ppp}.tif, where {ppp} goes from 0 to 791. I have successfully assembled the C1 serie resulting in a C1-global-positions-0.txt file. When I try to assemble the C2 serie using the previous metadata, I got the error above.

Am I doing something wrong ?

Alexandre

mmajurski, 2018-02-27:
Hello,

First I want to confirm some details. 
1) You stitched channel 1 (filename pattern: "C1-{ppp}.tif")
2) That stitching result was acceptable

If that is correct, my first guess would be: when you setup the assemble from metadata for channel 2, you want to indicate the global positions file generated by channel 1's stitching. However, you need to modify the filename pattern so MIST knows to stitch channel 2. 
So when assembling from metadata the filename pattern should read: "C2-{ppp}.tif". MIST only uses the position iterator {ppp} to look up in the global positions file which (x,y) position to use. 

If you are already modifying the filename pattern when using assemble from metadata, I am going to need to see the file "C1-global-positions-0.txt" to try to figure out why it is failing. The log file would also be useful.

Thanks, 
Michael

alexandrebastien, 2018-02-27:
Yes, I stitched C1-{ppp}.tif with a good result. I modified the file pattern to C2-{ppp}.tif for the second run.
[C1-global-positions-0.txt](https://github.com/usnistgov/MIST/files/1763666/C1-global-positions-0.txt)
[C1-log.txt](https://github.com/usnistgov/MIST/files/1763667/C1-log.txt)

mmajurski, 2018-02-27:
The problem is you are somewhere that uses a "," as the decimal separator as opposed to a "."

Java's ParseDouble does not respect the current locale. So when the global positions file got written it used the locale "," but tried to read expecting a "."

Update Fiji and let me know if that fixed your problem.

-mm

closed 2018-02-27 by mmajurski

alexandrebastien, 2018-02-27:
I actually append to have a french Win10 version since I live in Quebec, Canada. The "," is often a problem. The update corrected the problem. Thanks for the quick patch.