usnistgov/MIST #5: How to compile from source

duyck, 2018-03-26:
I want to edit stitching executor from the source code download and then recompile to make a new .jar file to be run as a plugin in ImageJ. 

The installation guide seems to be missing how to compile this package from source, only providing a link to .jar downloads.

Thank you.

mmajurski, 2018-03-26:
In theory, cloning the github repo and then using the maven pom.xml file should setup the project for you.

To rebuild the Fiji plugin jar I have an Artifact setup in IntelliJ which creates the plugin jar. However, the artifact just gathers the MIST module output and the required libs and packages them into the format expected by Fiji.
Here is the Artifact xml configuration.
https://gist.github.com/mmajurski/d226701784e06455b86a083ca22a0001

What IDE are you using to edit MIST?

Thanks,
-Michael

mmajurski, 2018-03-29:
When you compiled the Jar, did you include the plugins.config file?

That is how Fiji knows to add the jar to the plugin menu.
That file needs to be included in the jar file you generate and copy to the Fiji.app/plugins folder.

If the plugin is not showing up in the plugin menu that is the most likely reason.

-mm

mmajurski, 2018-03-29:
Excellent. 

I have done the same thing many times. 

-mm