usnistgov/MIST #7: Error

xiapunker, 2018-05-30:
Hello. I am trying to stitch together 35 images using row-column. I am trying to generate the preview (0%), but it gets to 92% and I get the error below. I have also tried to stitch using the sequential pattern and get the same error. I read through the manuals and could not figure out what the error was or how to fix it. Please help me. Thank you.

Devin

Exception in thread "Thread-19" java.lang.ArrayIndexOutOfBoundsException: 32251033
	at ij.process.ColorProcessor.set(ColorProcessor.java:313)
	at gov.nist.isg.mist.lib.export.blend.OverlayBlend.blend(OverlayBlend.java:89)
	at gov.nist.isg.mist.lib.export.LargeImageExporter.exportImageNoOverlap(LargeImageExporter.java:268)
	at gov.nist.isg.mist.lib.executor.StitchingExecutor.previewNoOverlap(StitchingExecutor.java:1030)
	at gov.nist.isg.mist.lib.executor.StitchingExecutor.run(StitchingExecutor.java:198)
	at java.lang.Thread.run(Thread.java:748)
Exception in thread "Thread-20" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "Thread-21" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "Thread-22" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "Thread-23" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "Thread-24" java.lang.ArrayIndexOutOfBoundsException

mmajurski, 2018-05-30:
Hello Devin,

Would you be willing to share both your input images as well as the params file so that I can recreate your error and dig into why it might be happening.

At first glance it appears to be an error when blending together the output stitched image. 

Also, I assume your input images are color, since the output stitched image exporter is using ImageJ's ColorProcessor to create the blended image?

Thanks,
Michael

xiapunker, 2018-05-30:
Hello Michael,

Please find the images attached (sorry for the change in file name, I tried
using FIJI just now, tiles are arranged top left to right and then down;
i.e. across r1, then r2, etc). The params file I don't have. It was nothing
special. I think I found another place on the site saying the files have to
all be the exact same size. This is likely my problem. Is there any way to
overcome this problem. Many organism images that need to be stitched
together do not have the same sizes.

Devin​
 tile_001.tif
<https://drive.google.com/file/d/1rHo9975mq1InAeZnnmg2IXW-stiSZ2Oc/view?usp=drive_web>
​​
 tile_002.tif
<https://drive.google.com/file/d/1YmsGnpXEkI0y2UIeScuKBP04XpdwEu5K/view?usp=drive_web>
​​
 tile_003.tif
<https://drive.google.com/file/d/1cPKLz46d4-nuIT-DMbhX-WScDPuxAbCO/view?usp=drive_web>
​​
 tile_004.tif
<https://drive.google.com/file/d/1qXrdOqKPcMug0NMKXL0J_uiWnSeeI-le/view?usp=drive_web>
​​
 tile_005.tif
<https://drive.google.com/file/d/1FeEjfFed1gmKNre2Qg97XV5xbpdGcmFS/view?usp=drive_web>
​​
 tile_006.tif
<https://drive.google.com/file/d/1saos51Jvfuwef3Jz_JPoWXfZzXPHyAuN/view?usp=drive_web>
​​
 tile_007.tif
<https://drive.google.com/file/d/1bOxmLJd7f7pQ0rSoJzRP8mIprScb9RyR/view?usp=drive_web>
​​
 tile_008.tif
<https://drive.google.com/file/d/1wyi3eiSHZsTMPVHR5jlPavofcovXwG5J/view?usp=drive_web>
​​
 tile_009.tif
<https://drive.google.com/file/d/1Eo0g6n3Ks1jNCjZ2kwIye9WSUPHt5CPf/view?usp=drive_web>
​​
 tile_010.tif
<https://drive.google.com/file/d/1MOsUwwOpN0djqA5kP3VxF69_Y5JxkQIq/view?usp=drive_web>
​​
 tile_011.tif
<https://drive.google.com/file/d/1qdvBRro3xMyxC1r3IEvkScTwf3-veWAp/view?usp=drive_web>
​​
 tile_012.tif
<https://drive.google.com/file/d/1Vg4Y2iYoNPALOUiwL2e2zW1Jk4wbMrkp/view?usp=drive_web>
​​
 tile_013.tif
<https://drive.google.com/file/d/1ShYZTe31o8x9fdZyXhVk5wujHz9atvq3/view?usp=drive_web>
​​
 tile_014.tif
<https://drive.google.com/file/d/1efNzEMw00DsZO3knGaO5aLlw7Fxnoegn/view?usp=drive_web>
​​
 tile_015.tif
<https://drive.google.com/file/d/11PJerfEu6l1yrJ_Cuyq2dkSXOfs6iWPV/view?usp=drive_web>
​​
 tile_016.tif
<https://drive.google.com/file/d/130N2xr8wR4nB8TcF9nWxcPXHNJuasRuv/view?usp=drive_web>
​​
 tile_017.tif
<https://drive.google.com/file/d/14f9zKFGzUWMGv4WDZoxKCKI1zP9B17DL/view?usp=drive_web>
​​
 tile_018.tif
<https://drive.google.com/file/d/1Vu3VgFPU76PHiZI-ae-Ple8ASbwknFGs/view?usp=drive_web>
​​
 tile_019.tif
<https://drive.google.com/file/d/1phKeraiJCrJrct9vV_KzTYbyEPPGnnnm/view?usp=drive_web>
​​
 tile_020.tif
<https://drive.google.com/file/d/1IJDESkbGBptddf6suF90DMRukbEp_w15/view?usp=drive_web>
​​
 tile_021.tif
<https://drive.google.com/file/d/1LmgJf7Hp1-f-xt_pAnLbHnHJI5IUMT6K/view?usp=drive_web>
​​
 tile_022.tif
<https://drive.google.com/file/d/1sQzbGzasKZ58So56bVbL310l9VncE5rm/view?usp=drive_web>
​​
 tile_023.tif
<https://drive.google.com/file/d/16Od-OC2gwO_Wv8cAIrIzKQcsGEWFCEVD/view?usp=drive_web>
​​
 tile_024.tif
<https://drive.google.com/file/d/11oq2BxOpgWFJvua0dDX-hxNa2Ywfea7e/view?usp=drive_web>
​​
 tile_025.tif
<https://drive.google.com/file/d/1fMfGKfdUjYP7xGYJ_gshW3yrcadfrNnp/view?usp=drive_web>
​​
 tile_026.tif
<https://drive.google.com/file/d/1x7EJXUhMEDLpfwvLYUB_8Tl_VSAIWC70/view?usp=drive_web>
​​
 tile_027.tif
<https://drive.google.com/file/d/1bTYFjkhX_NNLP9gHEgxNWLMmDCmIXhab/view?usp=drive_web>
​​
 tile_028.tif
<https://drive.google.com/file/d/1Fx2X4Gqi72hU6u3mnzYal2aYF4mdSSzb/view?usp=drive_web>
​​
 tile_029.tif
<https://drive.google.com/file/d/1-hoXChU3FqxP-NJDrl_-fWukgKZVxFL4/view?usp=drive_web>
​​
 tile_030.tif
<https://drive.google.com/file/d/1RtvCg0_G1rbAtIAtpcAhgxrIitAIQrt0/view?usp=drive_web>
​​
 tile_031.tif
<https://drive.google.com/file/d/1a27Ca4lxQZqqw303udwVtzjMdroPGkyX/view?usp=drive_web>
​​
 tile_032.tif
<https://drive.google.com/file/d/1MrIyzPIikag4F9U1pSuQIn3g2-NzmlHn/view?usp=drive_web>
​​
 tile_033.tif
<https://drive.google.com/file/d/1IQoCX6KOPurCfaBp4EZJh-FSabEg-i1b/view?usp=drive_web>
​​
 tile_034.tif
<https://drive.google.com/file/d/14KPr6le0MAWNSkFBPZFopwE-bKtWfF4S/view?usp=drive_web>
​​
 tile_035.tif
<https://drive.google.com/file/d/1ojw8ylhn4uFiLR_B_ibAn7mr4FiOhy0b/view?usp=drive_web>
​

mmajurski, 2018-05-31:
Devin,

MIST is expecting all the input images to be the same size (width, height). 
Without that MIST cannot stitching the images.

Out of curiosity, what microscope/camera system produced your images where the size is variable? 

And just FYI, to export parameters from MIST you configure everything in the UI and then click the "Save Params" button. That will generate a txt file with your current configuration. That file can then be loaded at a later point using "Load Params"

-Michael

closed 2018-05-31 by mmajurski

xiapunker, 2018-05-31:
Hello Michael,

I am using a Leica wide field microscope. Taking z stack images and
compressing into a single 2d image. I'm not sure exact microscope or camera.

Thank you for the help. I will move on to another stitching software.

Devin

mmajurski, 2018-05-31:
I would think that the camera would only produce a single size of image.

Does the Zstack produced by Leica have any x,y as you move down the stack thatLeica might be compensating for when it compresses them into a single 2D image? In other words, if the images making up the z stack have slightly different (x,y) stage positions by a few pixels, would Leica take that into account when building the composite max (or otherwise) projected image?

If you have the raw z stacks I would be interested in seeing if those images are all the same size, and whether a straight max projection (without any drift compensation) would be amenable to stitching.

Thanks,
Michael

xiapunker, 2018-05-31:
Hello Michael,

I will have to check the z-stack image sets. But when it compresses, yes
there is some slight variability in stage positions.

Devin

mmajurski, 2018-05-31:
During previous work with Leica Z stack stitching, I performed the max-projection operation myself, so the output image size was identical to the Leica camera. I have a guess that the differing image sizes is an artifact of how Leica compensates for (x.y) drift during Zstack compression into a single 2D image. 

If performing a max projection on the original stacks produces a visually acceptable image where all spatial tiles are the same width and height, then MIST should have no problem stitching your images.

-Michael