usnistgov/MIST #8: Problem with stitching histopathology slides

kvinicki, 2019-02-14:
Hi,

I'm working with canine cutaneous mast cell tumors. After couple of days working with MIST I still can't get satisfactory results and I really don't know what is the problem

I'm using "DIY" microscope slide scanner with Sony mirrorless a6300 camera and PLAN 20X microscope objective. Stepper motors are quite accurate (backlash is solved in the python code). I have a slight pincushion distortion, but even after correcting it with the [lensfunpy](https://github.com/letmaik/lensfunpy), results are not better. Horizontal overlap is ~29% and Vertical overlap is ~33% 

I uploaded the 28x10 image matrix: https://drive.google.com/open?id=14ccGfAu9SuiGXfzt_6462lFFT7bZ1Py2

mmajurski, 2019-02-14:
I am downloading the data now to see whats going on. 

Could you elaborate on "backlash is solved in the python code"? I am curious how you solved it.

Thanks,
Michael

mmajurski, 2019-02-14:
Could you also elaborate on the what the grid arrangement is? The files are names ######.jpg. Given the numbering I am assuming that it is row and column numbered with {rrr}{ccc}.jpg. 

If that is the correct interpretation of the filename, what is the width and height of the grid? 28 images tall, or 28 image wide?

Thanks

kvinicki, 2019-02-14:
Yes, the grid arrangement is {rrr}{ccc}.jpg (28 images wide, 10 images tall).

I think that I know what is the problem. The camera wasn't fixed properly and **maybe** they were some slight movement around the camera axis. I don't know how sensitive the algorithm is to this kind of movement? I will post an update after the microscope finishes with another scanning

About the backlash: I wrote a python script in which after every row a stepper motor that is controlling x-axis makes additional 56 steps in the other direction thus cancelling the backlash. Here is the code that I'm using (it is still quite messy): 
[20X_magnification_v0.6.zip](https://github.com/usnistgov/MIST/files/2866950/20X_magnification_v0.6.zip)
Is there maybe a better solution?

Also, I really like your work and I wouldn't be able to work on my project without MIST. I can provide you with additional data sets, if that would be useful to you in fine-tuning MIST algorithms :)

kvinicki, 2019-02-15:
Well, unfortunately that didn't solve the problem :(

mmajurski, 2019-02-25:
tl;dr, the vertical translations between your images violate MIST's assumption that the image overlap per direction is constant.

Now for the long version:

If you examine the translations for the vertical component of the image grid there are at least three overlap modes. One centered around 930 pixels (38% overlap), one centered near 980 pixels, and one centered around 1038 pixels (30% overlap). MIST requires a uniform grid where the approximate overlap between images is constant (per direction). For example, 30% overlap vertically with +-5% error. 

For the rest of this analysis I am relying on data generated from the MIST Logger at Log Level: INFO.

MIST determines which translations are valid based on the correlation between the overlapping image regions. Here is the set of valid translations used to compute the overlap. Once an overlap is computed, and translations which do not match the stage model are replaced based on the valid set of translations for that direction. 
`Translation used for MLE West:
[1548, 1563, 1552, 1608, 1624, 1620, 1665, 1638, 1636, 1591, 1594, 1618, 1585, 1597, 1624, 1523, 1581, 1536, 1545, 1642, 1608, 1606, 1650, 1627, 1602, 1607, 1631, 1629, 1590, 1611, 1622, 1521, 1546, 1560, 1550, 1609, 1624, 1609, 1665, 1632, 1633, 1597, 1596, 1616, 1582, 1599, 1622, 1516, 1587, 2254, 1577, 1540, 1542, 1646, 1609, 1595, 1651, 1609, 1596, 1608, 1626, 1622, 1589, 1608, 1622, 1520, 1543, 1555, 1552, 1605, 1617, 1612, 1664, 1627, 1628, 1593, 1595, 1615, 1581, 1598, 1617, 1514, 708, 1577, 1545, 1542, 1642, 1607, 1594, 1650, 1615, 1597, 1607, 1622, 1625, 1587, 1607, 1621, 1521, 1583, 1562, 1573, 1543, 1560, 1545, 1607, 1620, 1613, 1672, 1627, 1630, 1594, 1601, 1609, 1587, 1590, 1626, 1509, 1390, 1432, 1635, 1565, 1543, 1546, 1634, 1606, 1615, 1643, 1603, 1595, 1610, 1623, 1627, 1584, 1621, 1620, 1525, 304, 1537, 1553, 1627, 1575, 1542, 1560, 1550, 1606, 1615, 1609, 1668, 1627, 1633, 1587, 1596, 1615, 1582, 1592, 1619, 1515, 1584, 1503, 1443, 1546, 2069, 1568, 1553, 1535, 1649, 1613, 1592, 1663, 1598, 1594, 1613, 1623, 1620, 1590, 1606, 1627, 1518, 1587]`

These horizontal translations between images are nicely distributed, as shown in the histogram.
[image]

On the other hand, the North translations are drawn from many distributions. Here are the valid translations: 
`Translation used for MLE North:
[883, 882, 879, 879, 877, 874, 875, 880, 875, 871, 870, 871, 871, 869, 870, 869, 868, 984, 981, 982, 985, 986, 988, 991, 989, 984, 988, 991, 992, 993, 992, 994, 993, 995, 958, 938, 938, 938, 935, 935, 933, 929, 932, 933, 932, 930, 929, 927, 930, 928, 929, 929, 928, 989, 988, 988, 992, 993, 994, 997, 995, 994, 995, 997, 998, 999, 997, 999, 998, 997, 931, 934, 933, 930, 929, 927, 923, 925, 930, 929, 926, 926, 925, 927, 924, 925, 925, 935, 963, 1098, 1096, 1096, 1099, 1099, 1102, 1105, 1105, 1098, 1101, 1102, 1102, 1103, 1101, 1104, 1103, 1102, 1097, 1069, 945, 942, 944, 945, 942, 942, 941, 939, 938, 937, 936, 935, 935, 935, 936, 935, 936, 936, 974, 1030, 1032, 1030, 1029, 1032, 1031, 1032, 1035, 1037, 1037, 1038, 1040, 1040, 1040, 1038, 1039, 1039, 1038, 286, 1035, 1024, 993, 967, 957, 955, 957, 957, 953, 954, 952, 948, 949, 949, 948, 947, 946, 945, 947, 945, 945, 946, 947, 957, 987]`

which shows many modes in the histogram. 
[image]

There are three mitigation techniques (in the order I would suggest):
1) determine what caused the vertical stage translations to be grouped into about 5 overlaps spanning 869 to 1040 pixels and adjust the stage.

2) Leverage MISTs ability to specify how much uncertainty exists in the image overlap for the grid. Go into Advanced Parameters and specify Percent Overlap Uncertainty. For this wide of a range, it would need to be set to at least 8. This will allow MIST to consider vertical overlaps as valid for the median translation +- 8%, which should cover the whole spread of overlaps in the vertical translations.

This does not quite work, because MIST cannot seem to find enough translations per row to handle the differing overlaps. For this approach to work, there would need to be a few valid translations per row, so that when MIST replaces the invalid translations each row starts close enough to its actual overlap that the hill climbing refinement can converge.

3) After all the translation estimation and filtering, MIST perform a hill climbing on the cross correlation surface to refine the translations. If you expand that search range far enough MIST should be able to find the correct translations even if each row has a different overlap. This translates to using the Advanced parameter Stage Repeatability. Normally, repeatability is used to account for things like stage backlash and repeatability and is a single digit or small two digit number. In this case, if I give it 90 ~ ((1040 - 869) /2) to enable +- repeatability to cover the entire range of vertical translations the stitching takes almost forever, but does converge to a visually correct answer. 

I have included my stitching-params text file below so you can reproduce the MIST parameters I used.

> Stitching Run-time parameters:
gridWidth: 28
gridHeight: 10
startTileRow: 1
startTileCol: 0
imageDir: /home/mmajursk/Downloads/mist/issue2/imgs
filenamePattern: {rrr}{ccc}.jpg
filenamePatternType: ROWCOL
gridOrigin: UL
numberingPattern: HORIZONTALCOMBING
assembleFromMetadata: false
globalPositionsFile: 
startRow: 0
startCol: 0
extentWidth: 28
extentHeight: 10
timeSlices: 0
isTimeSlicesEnabled: false
[... cut ...]
outputImgPyramid: false
blendingMode: OVERLAY
blendingAlpha: NaN
outFilePrefix: img-
programType: AUTO
numCPUThreads: 40
loadFFTWPlan: true
saveFFTWPlan: true
stageRepeatability: 90
horizontalOverlap: NaN
verticalOverlap: NaN
numFFTPeaks: 0
overlapUncertainty: NaN
isUseDoublePrecision: false
isUseBioFormats: false
isSuppressModelWarningDialog: false
isEnableCudaExceptions: false
translationRefinementMethod: MULTI_POINT_HILL_CLIMB
numTranslationRefinementStartPoints: 16
headless: false
logLevel: INFO
debugLevel: NONE

closed 2019-02-25 by mmajurski