usnistgov/MIST #10: Not enough memory to create output stitched image.

Tommy7013, 2019-06-12:
Hi, 
  Something wrong to the MIST, when I want to stitch 5000 RGB pictures, I have enogh disk , swap and virtual memory. Could you help me to figure it out?

Thanks 
Tommy

tblattner, 2019-06-12:
Hi Tommy,

Unfortunately the current maximum number of pixels of an output image from the Java code is 2,147,483,647 pixels. I will pass this onto my colleague, who I believe has an external script that can be used to assemble larger images using the global position list.

Regards,
Tim

mmajurski, 2019-06-12:
If Matlab is a language you can use, I had a script for assembling arbitrarily sized images (so long as they fit in system memory).

I have not run this code in literally years, so its a coin flip as to whether it still works.

~Michael

[assemble_stitching.zip](https://github.com/usnistgov/MIST/files/3281573/assemble_stitching.zip)

Tommy7013, 2019-06-13:
Hi, 
   Got you. Thank you very much. It's really a big help.

Thanks 
Tommy

Tommy7013, 2019-06-13:
Hi,  
   Michael, thank you very much. It's really a big help. And thanks for you guys' open source.

 Best wishes
Thanks
 
Tommy

closed 2021-02-26 by mmajurski