usnistgov/MIST #12: Stitching 32-bit images results in a corrupter, white output image

tb-lund, 2019-08-21:
MIST is amazing and works really well is almost every case. However, I have an issue to get it to work on one of my workflows. In this workflow, I have to use 32-bit grayscale images saved from ImageJ into tiff files. When I generate a preview in MIST it shows this perfectly and when I perform the stitching the parameters it calculates (overlap and even global positions for each image) are all correct, meaning that it can interpret the images. However, when it blends them and outputs an image, this is just a white image with no retained information and when looking at the "levels", it appears that it is corrupted.

I have plenty of ram (196 GB, 175 allocated to imageJ) to memory should not be an issue even though the final image is 1Gb in size. I have tried this on MacOS and Ubuntu linux with the same result. Any tip would be highly appreciated.

kvinicki, 2020-05-04:
If you have global positions, you can use an external script to stitch the images.

tblattner, 2020-07-23:
Sorry to not get to this until now. The issue is how we store the pixel data for the full image. It uses the Java BufferedImage which is limited to 32-bit Integers. A major issue right now, that I am hoping to fix in the future. Its been a struggle to find cycles to work on it unfortunately.

I think at this time your best bet is to do as kvinicki says and write an external script to assemble the image using the resulting global positions file.

closed 2020-07-23 by tblattner