usnistgov/MIST #14: Unable to load FFTW

osoyer, 2020-07-22:
Running MIST with FFTW option produces "Unable to load FFTW" error. 

I installed FFTW as explained here; https://github.com/USNISTGOV/MIST/wiki/Install-Guide#fftw-installation

There were no issues during install. 

To try and troubleshoot, I checked that "libfftw3.dll" was under /Fiji.app/lib/fftw". However, there seem to be no "fftw" related files under "/usr/lib/". Hence I tried installing fftw again by running ">brew install fftw" on Terminal. This progressed without errors.

Any further help would be appreciated. As a side note, I can run MIST plugin with JAVA option.

best
orkun

tblattner, 2020-07-22:
Hello,

When you compiled FFTW and installed it. Can you identify the installation location of where the library was installed?

What operating system are you working with?

Regards,
Tim

osoyer, 2020-07-22:
Hi Tim,
I followed the instructions as on the above website, so I assume the "sudo make install" had installed it under "/usr/lib"...similarly for "brew" 

I can currently see a "fftw" folder under /Fiji.app/lib/" and under "/usr/local/Cellar/".

I'm on MacOSX Catalina (10.15.6).

best
orkun

tblattner, 2020-07-22:
Can you provide more details of the name of the library? And what option do you specify in the Advanced options for FFTW: https://github.com/USNISTGOV/MIST/wiki/User-Guide#fftw-execution

Specifically the "FFTW Library File".

I think if you select the exact FFTW library file (I think it should be /usr/lib/libfftw3.dylib, or equivalent)

osoyer, 2020-07-22:
I just tried the following: 
"/Applications/Fiji.app/lib/fftw/libfftw3.dll"
"/usr/local/Cellar/fftw/3.3.8_2/lib/libfftw3.dylib"

neither of these worked....I don't have any "fftw" files under /usr/lib/

tblattner, 2020-07-22:
The dll version is for Windows only, so that will not work.

Can you save your parameters and attach them?

Also can try: "/usr/local/Cellar/fftw/3.3.8_2/lib" as well as another option.

I am trying to remember if the library needs to be the 32-bit version or not. That could also be an issue here as well.

osoyer, 2020-07-22:
That path didn't work either. Parameter file attached.
[mist_prms1.txt](https://github.com/usnistgov/MIST/files/4961449/mist_prms1.txt)

tblattner, 2020-07-22:
Sorry, can you also upload the version with: "/usr/local/Cellar/fftw/3.3.8_2/lib/libfftw3.dylib" ?

Also, I just checked and we indeed do use isUseDoublePrecision: false by default. so the FFTW library will need to be compiled for single precision. I think the name may be libfftw3f.dylib.

To compile that version, just need to specify "--enable-float" when configuring. If this fixes the issue, I will update the wiki to reflect it.

osoyer, 2020-07-22:
Here it is....I did both options with "fftw3" and "fftw3f"
[mist_prms3.txt](https://github.com/usnistgov/MIST/files/4961511/mist_prms3.txt)

[mist_prms2.txt](https://github.com/usnistgov/MIST/files/4961503/mist_prms2.txt)

tblattner, 2020-07-22:
Hmmm, might want to try: ftp://ftp.fftw.org/pub/fftw/fftw-3.3.4.tar.gz

Apologies, it has been awhile since we have visited the version. Looking at the release notes for FFTW it looks like they may have added additional APIs. I'm wondering if this is conflicting with the JNI wrappers in some way or another... I will need to test on my end when i have some cycles to look at this in deeper detail.

I'm hoping by using the earlier version then it could work. Most likely using the fftwf version.

osoyer, 2020-07-22:
ok. Hopefully you will find a fix. I tried installing 3.3.4 as well - in the same way before. It didn't work, although I am not sure if I was correctly selecting it. Under 'usr/local/lib', there are now several files but none to distinguish as 3.3.4 or 3.3.8. I got there the following dylib files; "libfftw3.3", "libfftw3", "libfftw3f.3", and "libfftw3f"......I tried all of them I think

tblattner, 2020-07-22:
In either case you will want to specify the full path in the options as you did here: /usr/local/Cellar/fftw/3.3.8_2/lib/libfftw3f.dylib 

It has been awhile since I have tested on Mac. One thing that is a little more involved is to try to clone the repo and run MIST from source. This should give a more meaningful error to the one reported in Fiji. 

I am struggling to find cycles to debug this, and I currently do not have a Mac (it is currently on NIST campus, and given the current situation is not easy to get onto campus). So if you need this immediately, then that may be your best bet. 

Alternatively, if you have a Windows machine, could install Fiji on that and run it there. This should automatically work with FFTW as we ship the dll in the plugin.

closed 2021-02-26 by mmajurski