usnistgov/MIST #15: Time series stitching output is not time series

osoyer, 2020-07-23:
Running MIST on a time series data and failing to produce a time series output. It only outputs the final time point of the series as a stitched image. My MIST parameters are attached.
[mist_prms.txt](https://github.com/usnistgov/MIST/files/4964557/mist_prms.txt)

tblattner, 2020-07-23:
Could you also please upload a copy of your log file from the execution?

osoyer, 2020-07-23:
Here it is
[Log.txt](https://github.com/usnistgov/MIST/files/4966060/Log.txt)

tblattner, 2020-07-23:
Thanks for the files.

It appears you have set the "timeSlices" parameter to be 10. I am guessing you will want "1-10"

This guide: https://github.com/USNISTGOV/MIST/wiki/User-Guide#mist-graphical-user-interface has details on the parameter called "Timeslices"

osoyer, 2020-07-23:
That's great. Thanks much - my bad.

This has worked now, however, I find that some slices result in a better stitch than others. Is it possible to use the stitch info from one set of slices for the entire stack? That should also be more time efficient.

Help on this would be much appreciated. I can also share some sample files if you like.

tblattner, 2020-07-23:
If you have some stitching issues, one thing that could help is to experiment with is the advanced parameters in the app.

https://github.com/usnistgov/MIST/wiki/User-Guide#advanced-parameters

This can be used to specify the expected overlap. Stage repeatability is a variable we use during optimization, so setting this value to around 6 or 10 could help find the position.

A combination of those two may help improve accuracy in stitching. And lastly the overlap uncertainty attempts to find optimal coordinates +/- the calculate (or specified) overlap values.

Lastly, it may be possible to use existing stitching metadata with the "Assemble from metadata" option. https://github.com/usnistgov/MIST/wiki/User-Guide#mist-graphical-user-interface

This is typically used with multi-channel data, so it may be necessary to do this one-by-one for each timeslice.

osoyer, 2020-07-23:
Thanks. For the metadata route, I am assuming I can use the positions file that the MIST itself produces. In that case, should I be using the "global" or "relative" positions file? Running as above for example produces 10 of each of the "global", "relative" and "relative-no-optimisation" text files.

tblattner, 2020-07-23:
You will need to use the global positions... Keep in mind that if there the timeslices are independent and the camera/stage resets its position, that it will most likely have some uncertainty and most likely stitching errors.

Let me know how it goes!

closed 2020-07-23 by tblattner