usnistgov/MIST #16: "Could not find file" with missing grid images

yfukai, 2021-01-22:
I'm enjoying this great plugin, thanks! 
According to [the documentation](https://github.com/USNISTGOV/MIST/wiki/assets/mist-algorithm-documentation.pdf), the algorithm allows missing grid images. However, when I run the GUI without `image_r001_c001.tiff`, for example, it throws an error `Could not find file: .../image_r001_c001.tiff. Please check your image directory, file pattern, and start tile to ensure they match the files in your image directory.` and fails to stitch. When I changed the `Column Start Tile` value to avoid the missing images, the stitching succeeded. 
Is the missing grid image is supported in the GUI plugin?

tblattner, 2021-01-22:
Hello,

Indeed, it should operate with missing images as long it forms a connected graph of tiles. The only exception is with the first tile in the grid. We have validation checks on the GUI arguments for that first tile to verify the grid specifications and filename patterns are correct or not. Is the first tile always missing in your grids?

yfukai, 2021-01-23:
Thank you for your reply! 
> Is the first tile always missing in your grids?

Yes, we often take a tiled image of a circular dish bottom that lacks any corners (as in the attached example image), which may be a common practice. It might be useful for those cases if there is an option that enables us to skip the validation.

[image]

yfukai, 2021-01-25:
I found that [this line]( https://github.com/usnistgov/MIST/blob/78cdaaea04c2fb0ac694afca05d6bfea843ad597/src/main/java/gov/nist/isg/mist/gui/panels/inputTab/InputPanel.java#L691 ) would be responsible for this. If the building instruction is provided somewhere, I'm happy to make a pull request for the validation-skipping option, if appropriate.

tblattner, 2021-01-25:
I think adding a validtion-skipping option would be okay. Feel free to submit the pull request and we'll review it.

yfukai, 2021-07-09:
Hello, 
Rather took time, but I ended up with writing my own Python implementation following the MIST paper, which also works with that irregular shape (tested but still not documented). https://github.com/yfukai/m2stitch
the tentative package name is m2stitch -- a MIST-inspired microscope image stitching tool, but could I mention this library like that? Are there anything I should be careful about when I release the package? I'm looking forward to any comments. @tblattner

yfukai, 2021-07-27:
Got a (private) reply from @mmajurski, thanks! I'll keep this issue open just in case, but please close if appropreate.

closed 2023-06-30 by tblattner