usnistgov/MIST #17: Manual Installation doesn't work

cvhorie, 2021-01-29:
The instructions are
Install Fiji, did that toC:\Users\****\Documents\fiji-win64\Fiji.app
"Manual Installation
    Download the latest version of the MIST plugin from MIST Plugin Download
    Extract the contents of the zip file into your Fiji.app folder
    Restart Fiji
    Plugin is installed in Plugins >> Stitching >> MIST"
Did that to C:\Users\****\Documents\fiji-win64\Fiji.app\MIST-master and then to C:\Users\****\Documents\fiji-win64\Fiji.app\ , but MIST is not listed in Stitching.
When I start Fiji (ImageJ-win64), I get a load of plugin configuration errors for: Bigwarp, Bio-Formats and N5 Viewer.

mmajurski, 2021-02-26:
When you unzip the MIST zip file, do you see a set of folders:
'jars'
'lib'
'plugins'

The contents of each of these folders should be added to each matching folder under the Fiji.app/ folder.

kabeero, 2022-05-10:
FYI: the link in the Manual Installation section gives 404
> https://github.com/USNISTGOV/MIST/wiki/Install-Guide#manual-installation

closed 2023-01-25 by tblattner