usnistgov/MIST #18: Stitching not working correctly

osoyer, 2021-02-15:
Dear MIST developers,
I'm trying to stitch a time-lapse taken over 9 positions organised in a grid around a central point with 20% overlap (horizontal and vertical) among all images. As a trial, I'm trying to stitch the first time point only. This does not work really. I attach the images for your attention. Would appreciate any help you can provide.

best
orkun
[ch1_grid.zip](https://github.com/usnistgov/MIST/files/5978862/ch1_grid.zip)

mmajurski, 2021-02-26:
Looking at your images, there appears to be a very large vertical shift between pairs of horizontal tiles. It looks like any pair of images is offset by 33 % vertically.

[image]
 

I cannot seem to find a set of MIST parameters which allows it to figure out this baked in vertical offset. 

MIST is expecting a regular grid of images. Some camera angle (which causes this kind of vertical shift between horizontal images) is expected and can be accounted for. But I have never seen an offset this large, and it appears that MIST cannot find the orthogonal translation (i.e. the vertical component of the translation between a left-right pair of images) that you have in this dataset. I am used to this vertical offset between horizontal images being in the range of 10s of pixels.

osoyer, 2021-02-28:
Thanks for looking into this @mmajurski . Indeed, since posting, I have remedied the issue with the microscope and am now getting better images in terms of overlap across a row. I will give it another try with MIST.

best
orkun

osoyer, 2021-03-08:
Dear @mmajurski, as I mentioned in my previous comment, here is a better aligned grid images. 

I attach the sample images from a single row with 3 columns. The vertical overlap is almost perfect and the horizontal overlap was 25% (as used on uManager grid creation facility). I'm still having hard time stitching these.....Below is the macro I use. What I get is a good stitch between column 1 and 2, but a loss of part of image between column 2 and 3.

file = "Ch1_r{rrr}_c{ccc}_t{ttt}.tif";
outfile = "Ch1_";
nslices = 1;
nrows=1;
ncols=3;
stagerep=1; //stage repeatibility. higher it is higher the stage translation error
hoverlap=75;	//expected horizontal overlap among positions - in percent
voverlap=99; //expected vertical overlap among positions - in percent
uoverlap=1; //expected overlap error - in percent

//call MIST
run("MIST", 
"gridwidth="+ncols+" gridheight="+nrows+" starttilerow=0 starttilecol=0 "+
"imagedir="+dir+" filenamepattern="+file+" filenamepatterntype=ROWCOL "+
"gridorigin=UL assemblefrommetadata=false "+
"assemblenooverlap=false globalpositionsfile=[] numberingpattern=HORIZONTALCOMBING startrow=0 startcol=0 "+
"extentwidth="+ncols+" extentheight="+nrows+" timeslices=0-"+nslices+" istimeslicesenabled=true "+
"outputpath="+outDir+" displaystitching=true outputfullimage=false "+
"outputmeta=true outputimgpyramid=false blendingmode=OVERLAY blendingalpha=NaN outfileprefix="+outfile+" programtype=JAVA "+
"numcputhreads=8 loadfftwplan=true savefftwplan=true fftwplantype=MEASURE fftwlibraryname=libfftw3 "+
"fftwlibraryfilename=libfftw3.dll planpath=/Applications/Fiji.app/lib/fftw/fftPlans "+
"fftwlibrarypath=/Applications/Fiji.app/lib/fftw "+
"stagerepeatability="+stagerep+" horizontaloverlap="+hoverlap+" verticaloverlap="+voverlap+" numfftpeaks=0 "+
"overlapuncertainty="+uoverlap+" isusedoubleprecision=false isusebioformats=false "+
"issuppressmodelwarningdialog=false isenablecudaexceptions=false translationrefinementmethod=MULTI_POINT_HILL_CLIMB "+
"numtranslationrefinementstartpoints=16 headless=false loglevel=MANDATORY debuglevel=NONE");

//close windows
windowlist = getList("image.titles");
for (j=0; j<windowlist.length; j++) {
	filename = outDir+outfile+"stitched_"+j+".tif";
	selectWindow(windowlist[j]);
	run("Canvas Size...", "width=6000 height=1250 position=Center zero");
	saveAs("Tiff", filename);
	close();
}
[sampleImages.zip](https://github.com/usnistgov/MIST/files/6102938/sampleImages.zip)

mmajurski, 2021-03-09:
All of the sampleImages not in the first row are pure black? Is that correct?

If I just run MIST on the first row, I get what looks to be a perfectly viable stitching result with the default parameters.

> System information:
Java Version: 1.8.0_172
Operating System: Linux amd64 v5.8.0-44-generic
CPU threads used: 6
Free memory available to JVM (GB): 0
Total memory available to JVM (GB): 0
Max memory for JVM (GB): 18
Execution type: Auto

>Execution timing and general information:
Statistics Output Version: 1.0

>Total time for experiment (ms): 3434
Time slice: 0
Total Stitching Time (ms): 3239
Relative Displacement Time (ms): 2919
Global Optimization Time (ms): 316
Stage Model Build Time (ms): 41
Global Position Time (ms): 1
Running sequential version (LOW MEMORY): false
Keep all pixel data in memory: true

>North repeatability: 0
North overlap: 22.0
North min filter threshold: 900.0
North max filter threshold: 972.0

>West repeatability: 0
West overlap: 53.0
West min filter threshold: 844.8
West max filter threshold: 960.0

>Error report: PASSED

>Stitching Run-time parameters:
gridWidth: 3
gridHeight: 1
startTileRow: 0
startTileCol: 0
imageDir: /home/mmajurski/Downloads/sampleImages
filenamePattern: Ch1_r00{r}_c00{c}_t00{t}.tif
filenamePatternType: ROWCOL
gridOrigin: UL
numberingPattern: HORIZONTALCOMBING
assembleFromMetadata: false
assembleNoOverlap: false
globalPositionsFile: 
startRow: 0
startCol: 0
extentWidth: 3
extentHeight: 1
timeSlices: 0
isTimeSlicesEnabled: true
outputPath: /home/mmajurski/Downloads/sampleImages
displayStitching: true
outputFullImage: false
outputMeta: true
outputImgPyramid: false
blendingMode: OVERLAY
blendingAlpha: NaN
outFilePrefix: img-
programType: AUTO
numCPUThreads: 6
loadFFTWPlan: true
saveFFTWPlan: true
fftwPlanType: MEASURE
fftwLibraryName: libfftw3f
fftwLibraryFilename: libfftw3f.dll
planPath: /home/mmajurski/programs/Fiji.app/lib/fftw/fftPlans
fftwLibraryPath: /home/mmajurski/programs/Fiji.app/lib/fftw
stageRepeatability: 0
horizontalOverlap: NaN
verticalOverlap: NaN
numFFTPeaks: 0
overlapUncertainty: NaN
isUseDoublePrecision: false
isUseBioFormats: false
isSuppressModelWarningDialog: false
isEnableCudaExceptions: false
translationRefinementMethod: MULTI_POINT_HILL_CLIMB
numTranslationRefinementStartPoints: 8
headless: false
logLevel: INFO
debugLevel: NONE

osoyer, 2021-03-10:
Hi @mmajurski . Indeed, you are right. The stitching seems working well with your parameters. I checked against what I had, and it seems a key difference is that you didn't specify horizontal/vertical overlap. I was using 0-5 for vertical and 25 for horizontal (which is what I had used on uManager)....So, is the best approach to leave these parameters undefined and let the algorithm work this out?

Also, I have over 200 timeslices - in this particular case - would it be better / faster to stitch one timeslice, record the stitching and then apply this to the remaining slices? How would I do that? Currently, I'm stitching each slice separately.

PS: I don't get your comment about sample images being black. You have applied 'auto contrast', yes?

tblattner, 2021-03-10:
Mike, correct me if I'm wrong, but from the earlier message @osoyer, you were using these parameters for the horizontal/vertical overlaps?
hoverlap=75; //expected horizontal overlap among positions - in percent
voverlap=99; //expected vertical overlap among positions - in percent

If the grid had 0-5 vertical and 25 horizontal, then these values should also mimic those:
hoverlap=25; //expected horizontal overlap among positions - in percent
voverlap=1; //expected vertical overlap among positions - in percent

These options are useful if our global optimization step fails to find good candidates for estimating the horizontal/vertical overlaps. A good candidate might be two neighboring images with features that may be used correlate the two. If there is insufficient content between tiles across the entire image, then specifying these values can assist the algorithm. Given there is sufficient content in the overlap areas of your tiles, then we were able to estimate the overlaps effectively.

mmajurski, 2021-03-10:
@osoyer Yes, I did auto contrast the images, all pixels were value 0 outside of the first row.
Here is time slice 0
[image]

@tblattner Correct, the MIST grid parameters should mimic the parameters used in constructing the grid in MicroManager. 
@osoyer The overlaps you talk about 
hoverlap=75; //expected horizontal overlap among positions - in percent
voverlap=99; //expected vertical overlap among positions - in percent
make little sense. If you have a horizontal overlap of 75%, then in a 1000 pixel image, your images would be offset by 250 pixels. Which is much less than the visual overlap seen in the images you shared. And a vertical overlap of 99, would almost completely overlap the images.

If you have time points there are two options:
1) specify the timepoint iterator and allow MIST to stitch each time point independently. 
https://github.com/usnistgov/MIST/wiki/FAQ#stitching-time-slices
2) stitching timepoint 0 and apply those translations to all other time points. If you want to do this, you will need to duplicate the global-positions file once per time point you want to stitch since "If your Filename Pattern has a time slice iterator "{tt}" then the global-positions file specified must also." from https://github.com/usnistgov/MIST/wiki/Combined-User-Install-Guide#input-parameters

osoyer, 2021-03-11:
Thank you both, @mmajurski and @tblattner. My overlaps are 25% and 1%. I tried these values and when my stitch didn't work, I tried 75% and 99% - in case I was interpreting the parameters incorrectly. Thanks for confirming the correct usage - for my images it should be 25% and 1%. 

I get a good stitch now without specifying the overlap parameters. I will continue playing with this and adapting to my time lapses.

PS: the images are fine on my end. I wonder if zip created the issue.

closed 2021-03-11 by mmajurski