usnistgov/MIST #19: Unsufficient stitching of SEM rat brain images

spezialekg, 2021-02-23:
Dear MIST developers,
I'm working with serial block face sem multi tile images of rat brain and am struggling with the MIST stitch application. It seems to apply a too large overlap in my stitched image. The images have been acquired as a 3x1 grid with a 10% vertical overlap in the stitch area. I've uploaded one set of these images. Due to size limitations I had to downsize the images (6115x6126pxls to 2000x2004pxls) and I see that the downsized stitched image is of even larger overlap or something than the original size stitched image. 
I hope to hear from you soon and that you have any suggestions for me how to tweak the settings of the MIST application. Please let me if you would like me to share the original images somehow.

Looking forward to hear from you,
Cheers, 
Karin
      
[3x1_stitch.zip](https://github.com/usnistgov/MIST/files/6031852/3x1_stitch.zip)

mmajurski, 2021-02-26:
If MIST is overestimating the overlap in any direction, you can specify what image overlap you are expecting using the advanced parameter tab.

See the wiki advanced parameters description:
https://github.com/usnistgov/MIST/wiki/Combined-User-Install-Guide#advanced-parameters

Try setting the vertical overlap to your known value and see what MIST produces.

spezialekg, 2021-03-15:
Dear mmajurski,
Thank you for your reply. I'm sorry I didn't specify this in my query, but as mentioned above my images have been acquired with a 10% vertical overlap and I did specify a 10% vertical overlap using the advanced parameter tab in MIST. However it seems as if MIST is not able to apply the correct estimation of overlap.

mmajurski, 2021-03-15:
What is the stage repeatability that MIST is calculating? If you tell MIST you have x% overlap, it will use that since any user supplied value is trusted. It will still compute the overlap, but it overrides the computed value with the user specified value. 

Might the stage repeatability be ending up too large, which allows the translations to drift +- stage repeatability from the specified overlap?