usnistgov/MIST #20: distribution of valid and invalid translations

captainst, 2021-03-02:
In the MIST algorithm documentation, there is an assumption: **"The subset of valid translations is expected to be approximately normally distributed. The subset of invalid translations is expected to be approximately uniformly distributed."**

So why are the valid translations are normally distributed while invalid translations are not ?

Many thanks !

kvinicki, 2021-03-04:
I'm also interested in this

mmajurski, 2021-03-04:
MIST needs to be able to figure out which translations it can trust to estimate the microscope stage properties, in order to reassemble the grid (more specifically, to determine what the search bounds are for the normalized cross correlation hill climb search which happens for each pairwise translation).

To do this MIST has a stage model which makes the above assumption that you mentioned. So given a vector of translations, the intuition is that the correct ones will be group around the ideal (theoretical) correct offset in the absence of any stage jitter and noise. The incorrect translations are the results of not enough information, or other secondary FFT peaks. So those translations will be more broadly distributed across the whole range of possible translations.

This leads to a clumped distribution (correct translations), and a uniform distribution (incorrect translations). While they are not likely to actually be normal or uniform, those distributions are general enough that the Stage Model can perform Maximum Likelihood Estimation to optimize the three parameters (probability of belonging to the uniform distribution, normal distribution mean, and normal distribution sigma).

In other words, the uniform distribution contributes a flat probability that any given translation is incorrect. While the normal distribution of translations contributes a mu and sigma (mean and standard deviation) of the grouped translations.

From experience, when the FFT translation estimation is working correctly on a well aligned image grid, the normal distribution parameters are very tight, the mean is often exactly correct, with a sigma roughly 1-2% of the relevant image dimension. 

If you would like to explore this further, plot the pairwise translations in each direction (i.e. plot the horizontal pair-wise translations) and look at the resulting histogram/distribution.

captainst, 2021-03-05:
@mmajurski Thank you very much! That is a very comprehensive introduction! As you pointed out, for valid translations, the sigma would be very small (sometimes <<1~2% of the relevant image dimension). In our experiments with 100x magnification, a few pixels (5) mis-alignment is noticeable for some parts, like sperm tails.

mmajurski, 2021-03-05:
The small mis-alignment is handled by the final translation optimization, which runs a hill climbing operation (with normalized cross correlation as the metric between two images). 

This hill climb is bounded by expected translation +- 2 x Stage Repeatability in either direction.
This hill climb maximizes the pair-wise normalized cross correlation to improve the local translation agreement. It however does not affect the rough overlap calculated earlier by the Stage Model.

If you need additional accuracy in this final translation refinement, you should adjust the [advanced parameter](https://github.com/usnistgov/MIST/wiki/Combined-User-Install-Guide#advanced-parameters) which defines the number of hill climbs (or even if you want to exhaustively compute all possible normalized cross correlation values within the translation refinement box of +-repeatability).

captainst, 2021-03-06:
@mmajurski thank you again! We have been using traditional feature-based stitching approach using motorized stages. The sequence of the tiles are known and pre-determined so that simplified a lot the process. By feature matching approach, if the matchings are reliable, then the stitchings also also good. We used pyramid and simplified RANSAC to achieve that.
By chance I came across your brilliant work (paper) and it is a new thought. I'll try the software. Many thanks.

closed 2021-03-08 by mmajurski