usnistgov/MIST #21: Stitching images that do not fully fill a grid

lambdamoses, 2021-04-16:
I'm trying to stitch some images taken just to cover a mouse brain without imaging empty spaces without tissue. The FOVs are supposed to be arranged like this:

[image]

So it sort of is vertical continuous, but not all cells in the grid has an image. The file names have the FOV index as shown. I can't get MIST to recognize this pattern. I tried to rename the files to have rows and columns, but MIST demands that all cells in the grid is filled and throws an error of not finding a file that isn't supposed to exist in the first place.

tblattner, 2021-04-16:
Unfortunately we currently expect a complete grid of tiles in order to stitch. At one point we did have a way of doing this in a  Matlab prototype. @mmajurski do you know if this feature is still in there or not?

I looked around the MIST plugin to see if it would be possible to add a custom grid, but I fear it will be a lot of effort to implement. If you have any developers, we'd be happy to provide insights into the program to help with such a feature. Mike and I have been discussing the possibility of creating a python package that follows the MIST algorithm. We will keep this use case in mind when we get the cycles to develop it. Right now we do not have those cycles though!

mmajurski, 2021-04-16:
MIST should support missing images, so long as the first image in the grid exists.
I.e. if your grid starts in the upper left for numbering, then MIST needs that first upper left image to figure out whether you started counting from 0 or 1. 

If the first image exists MIST should work. However there is an additional condition: there needs to be only one connected tree of images. MIST using minimum spanning tree to select which translations to use to build the image. So there can only be a single spanning tree in the grid of images. This translates to no non-connected groups of images in the grid. To clarify, when I say connected, I mean adjacent with a horizontal or vertical translation. So any image which is 4 connected to the larger set of images.

One hack you can try is to create blank images to fill in the first row until it connects to the grid of images you described above. 

I don't completely recall, but for these tricky types of image grids, its almost always better to use row-column numbering instead of sequential numbering. In fact, that _might_ be a requirement of the missing tile stitching functionality.

yfukai, 2021-09-09:
Hi there, this might be related to issue #16, and I wrote a Python package that can treat those tiles without corners. Please check https://github.com/yfukai/m2stitch/ and it can be useful hopefully.