usnistgov/MIST #22: Enable to save the stitched image  with Pyimagej

tusharvora, 2021-05-17:
I am using the python Pyimagej library to run the MIST plugin. I am able to get all the files (for. eg -global position file, relative position file) but enable to save the Stitched image even after keeping 'outputfullimage: True'. Is there anything I am missing?

Image statistics file shows 'outputfullimage:' argument false after stitching.

`# -*- coding: utf-8 -*-

import os
import imagej

ij = imagej.init('C:/Users/tushar/Desktop/Fiji.app', headless=False)

# Run Plugin
plugin = 'MIST'

args = { 
    'gridwidth': 7,
    'gridheight': 9,
    'starttileRow': 0,
    'starttileCol': 0,
    'imagedir': 'C:/Users/tushar/Desktop/supersonic/PCB_Matching/pad_mask/stiching_testing/test1/images/20',
    'filenamepattern': 'r{r}-c{c}.jpg',
    'filenamepatterntype': 'ROWCOL',
    'gridorigin': 'LL',
    'numberingpattern': 'HORIZONTALCOMBING',
    'assemblefrommetadata': True,
    'assemblenooverlap': False,
    'globalpositionsfile':'C:/Users/tushar/Desktop/supersonic/PCB_Matching/pad_mask/stiching_testing/test1/images/20/img-global-positions-0.txt',
    'startrow': 0,
    'startcol': 0,
    'extentwidth': 7,
    'extentheight': 9,
    'timeslices': 0,
    'istimeslicesenabled': False,
    'outputpath':'C:/Users/tushar/Desktop/supersonic/PCB_Matching/pad_mask/stiching_testing/test1/images/20',
    'displaystitching': True,
    'outputfullimage': True,
    'outputmeta': True,
    'outputimgpyramid': False,
    'blendingmode': 'LINEAR',
    'blendingalpha': 'NaN',
    'outFileprefix': 'img-',
    'programtype': 'AUTO',
    'numcputhreads': 8,
    'loadfftwplan': True,
    'savefftwplan': True,
    'fftwplantype': 'MEASURE',
    'fftwlibraryname': 'libfftw3f',
    'fftwlibraryfilename': 'libfftw3f.dll',
    'planpath': 'C:/Users/tushar/Desktop/Fiji.app/lib/fftw/fftPlans',
    'fftwlibrarypath': 'C:/Users/tushar/Desktop/Fiji.app/lib/fftw',
    'stagerepeatability': 0,
    'horizontaloverlap': 'NaN',
    'verticaloverlap': 'NaN',
    'numfftpeaks': 0,
    'overlapuncertainty': 'NaN',
    'isusedoubleprecision': False,
    'isusebioformats': False,
    'issuppressmodelwarningdialog': False,
    'isenablecudaexceptions': False,
    'translationrefinementmethod': 'EXHAUSTIVE',
    'numtranslationrefinementstartpoints': 16,
    'headless': False,
    'loglevel': 'MANDATORY',
    'debuglevel': 'MANDATORY'
}
ij.py.run_plugin(plugin, args)
`

[img-global-positions-0.txt](https://github.com/usnistgov/MIST/files/6495459/img-global-positions-0.txt)

[img-statistics.txt](https://github.com/usnistgov/MIST/files/6495440/img-statistics.txt)
[img-log.txt](https://github.com/usnistgov/MIST/files/6495443/img-log.txt)

tblattner, 2022-05-06:
A new version of MIST has been pushed that should support arbitrary sized images.

closed 2022-05-06 by tblattner

sqUadow, 2024-02-28:
For anyone else encountering this issue, boolean arguments need to be passed as lower case strings. This worked for me:

`plugin = 'MIST'

args = {
'gridwidth': 2,
'gridheight': 2,
'starttile': 1,
'imagedir': 'D:\stitch test\plate_04\XY01\T0001',
'filenamepattern': 'test_capture_smaller_grid_T0001_XY01_0000{p}_Z001_CH3.tif',
'filenamepatterntype': 'SEQUENTIAL',
'gridorigin': 'UL',
'numberingpattern': 'HORIZONTALCONTINUOUS',
'assemblefrommetadata': 'false',
'assemblenooverlap': 'false',
'globalpositionsfile':'',
'startrow': 0,
'startcol': 0,
'extentwidth': 2,
'extentheight': 2, 
'timeslices': 0,
'istimeslicesenabled': 'false',
'outputpath':'D:\stitch test\plate_04\XY01\python_test',
'displaystitching': 'true',
'outputfullimage': 'true',
'outputmeta': 'true',
'outputimgpyramid': 'false',
'blendingmode': 'OVERLAY',
'blendingalpha': 'NaN',
'compressionMode': 'UNCOMPRESSED',
'outFileprefix': 'img',
'unit': 'MICROMETER',
'unitX': 1.0,
'unitY': 1.0,
'programtype': 'AUTO',
'numcputhreads': 12,
'loadfftwplan': 'true',
'savefftwplan': 'true',
'fftwlibraryname': 'libfftw3f',
'fftwlibraryfilename': 'libfftw3f.dll',
'planpath': r'C:\Users\Paul\Desktop\Fiji.app\lib\fftw\fftPlans',
'fftwlibrarypath': r'C:\Paul\me\Desktop\Fiji.app\lib\fftw',
'stagerepeatability': 0,
'horizontaloverlap': 30.0,
'verticaloverlap': 20.0, 
'numfftpeaks': 0,
'overlapuncertainty': 20.0,
'isusedoubleprecision': 'true',
'isusebioformats': 'false',
'issuppressmodelwarningdialog': 'true',
'isenablecudaexceptions': 'false',
'translationrefinementmethod': 'EXHAUSTIVE',
'numtranslationrefinementstartpoints': 16,
'headless': 'false',
'loglevel': 'MANDATORY',
'debuglevel': 'NONE'
}
ij.py.run_plugin(plugin, args)`