usnistgov/MIST #23: Scaling for Very Large Images

AcidLeroy, 2021-07-25:
I was curious if anyone in this community has had experience running MIST across multiple machines? I.e is it possible to have multiple sections that are composed of tiles that need to be stitched across machines? Let me explain below: 

Overall Image
+----------+------------+
|                   |                      |
|        A         |           B        |
|                   |                      |                  
+----------+----------- +
|                   |                      |
|         C       |            D        |
|                   |                      | 
+----------+------------+

Let's say that I have an overall image like the above. A, B, C, and D are arbitrary splits of the overall image. Each subsection, A, B, C, and D is all composed of tiles that need to be stitched. Normally, I would run MIST over the entire image (I wouldn't split the image into subsections). But if I want to parallelize A, B, C, and D over multiple machines, is this possible? If I want Machine A to compute subsection A, Machine B to compute subsection B, etc, is there an option to do this? 

Thank you!

AcidLeroy, 2021-07-27:
@mmajurski thank you very much for the response! Are you aware of any algorithms that can do this over multiple machines? What would be the issue be if I used MIST on my subsections A, B, C, and D, and then treated A, B, C, and D as tiles to be stitched together to create the overall image? This would be a "two-pass" approach. I would imagine that the issue might be that the newly generated tiles would be too big to actually stitch. I'm assuming that the entire tile must fit into memory in order to perform the computation?

mmajurski, 2021-07-27:
I am not aware of any algorithm like that.

Can I ask why you are trying to split the work across 4 ndoes? Is there a runtime constraint? There should not be a memory constraint since MIST can stitch as long as it has enough memory to hold a pair of images (and the FFTs).

Does MIST not work if you just give it the whole grid of images to stitch together? If not, what errors is it giving you?

AcidLeroy, 2021-07-27:
Ok, thanks. 

Four nodes are just an example, but I am interested in horizontally scaling these stitching operations. Right now we scale by getting bigger and better machines, but I want to see if there is a way to implement this over N-nodes. There are currently no errors, but for some of our smaller operations, the algorithm takes on the order of hours and can take much longer for our bigger jobs. I was interested in reducing this time by splitting it over multiple GPU or CPU machines. I.e. can I speed this process up if I have 10k machines available to me?

mmajurski, 2021-07-27:
What size of images are you stitching that MIST takes hours?

Using the FFTW version on modern desktop hardware usually approaches the I/O speed of the storage medium. In the past I have certainly bottlenecked a spinning hard disk.

closed 2023-01-25 by tblattner