usnistgov/MIST #24: Does it support sub-pixel registration?

Hubrt-Zhu, 2021-12-06:
I found that the positions of all images in the img-outglobal-positions-0.txt file are integers.
I would like to ask if it is possible to choose sub-pixel registration like Fiji.

ibrahim-phomedics, 2024-10-09:
Hey @mmajurski, if I was looking to build sub-pixel resolution capability into MIST, any ideas on how to do so?

mmajurski, 2024-10-09:
If you wanted to add in sub-pixel resolution I would leave the majority of the algorithm alone, and once its done building alignments between images, take those alignments and refine them to your sub-pixel resolution. 

The algo has 2 broad states, pair-wise translation rough approximation, and then pixel-wise refinement. The first uses PCIAM, the second a multi-point hill climb in normalized cross correlation space. Once the pixel-wise refinement is complete, you could add in another refinement to sub-pixel. But doing so will require you handle the image assembly slightly differently (the minimum spanning tree for assembly should work) but the actual pasting of tiles into the output would likely fail. So that would need to be updated to support sub-pixel.

ibrahim-phomedics, 2024-10-10:
@mmajurski, great thanks! I managed to get it working with a python port of MIST and [phase_cross_correlation](https://scikit-image.org/docs/stable/api/skimage.registration.html#skimage.registration.phase_cross_correlation).