usnistgov/MIST #25: CUDA 10 & 11 support

xixixihaha, 2022-01-19:
CUDA 9.0 in not officially supported on Ubuntu 18.04 LTS and Ubuntu 20.04 LTS.
Is it possible to migrate the CUDA code to support the latest version?

kabeero, 2022-02-09:
Hi,

There is a section on the Wiki about rebuilding the project with [certain](https://mvnrepository.com/artifact/org.jcuda/jcuda) newer CUDA versions:
> https://github.com/USNISTGOV/MIST/wiki/Install-Guide#how-to-modify-mist-to-use-an-older-or-newer-version-of-cuda

However, I see ~5 months ago developer dropped support for CUDA altogether:
> https://github.com/usnistgov/MIST/commit/8d8ac4846a69584af232ab73ce9ceb955cb002ff

:(

tblattner, 2022-02-09:
Unfortunately, since the Volta GPU architecture some of the CUDA code that used implicit synchronization would no longer be reproducible. 

Because of this we have dropped GPU support. I do have adding support back in on my back log, but have not managed to find the cycles to reimplement the CUDA kernels.

If others are interested in helping, here are the relevant CUDA kernels.
https://github.com/usnistgov/MIST/blob/master/lib/jcuda/util-cuda-bin.cu

The kernels that would need to be fixed are the reduce_max_final, reduce_max_main, reduce_max_filter_final, reduce_max_filter_main, reduce_max_finalf, reduce_max_mainf, reduce_max_filter_finalf, and reduce_max_filter_mainf.

The 'f' versions are for single precision.

The functionality of these functions are parallel reductions, so fairly common. The only caveat is the 'reduce_max_filter' functions which finds multiple max values where the indices of these max values are a certain distance from one another. We could do a radix sort for this potentially if we can capture the original indices alongside the sorted array.

I did find the 'CUB' library that could provide the needed functionality: https://nvlabs.github.io/cub/

Which has parallel reductions and sorts available. Alternatively, could possibly be reimplemented using Thrust.

I'm curious what type of computers you are using for stitching. We hadn't had a strong use case for the  GPU version as the disk IO often ended up being the bottleneck, and the CPU was able to keep up with the disk. I could envision very high resolution tiles could justify the need for the GPU acceleration. Most images we stitched were on the order of 1392 x 1040.

kabeero, 2022-02-10:
Thanks @tblattner, appreciate your detailed reply.

We are working with 16 MP microscope photos in 1-3K batches. We originally intended to use ImageJ's stitching but wanted to accelerate the processing time. If we were to experience disk limitations, we were going to use a ramdisk coming out of the microscope software.

The other candidate we are looking into is [OpenCV's Stitcher with GPU](https://docs.opencv.org/3.4/d8/d19/tutorial_stitcher.html).

I don't believe I have the knowledge to reimplement the CUDA code, thanks for sharing the ideas.

kabeero, 2022-05-06:
Thanks @tblattner, indeed it is quite fast and CUDA may not be needed. I am able to stitch ~1500 microscope photos in <3 minutes with your algorithm.

When I scale up, I seem to be getting a lot of the Out of Memory issues as described in your FAQ... Out of curiosity, are there any non-Java programs you have seen work well to do the final placement + blending?

tblattner, 2022-05-06:
Just spoke with MIke. Here is the assembly script.

https://github.com/usnistgov/MIST/blob/master/assemble_stitched_image.py

In the meantime we are finalizing a code review to incorporate the new methodology for exporting the stitched image. This version should scale to arbitrary sized images. Once the code review is done we will upload it to the Fiji update site. So should be as simple as updating the plugin from Fiji to get it working.

tblattner, 2022-05-06:
Good news @kabeero. We've updated to the latest version, and it should now support exporting arbitrary sized images. Let me know how it goes!

closed 2023-01-25 by tblattner