usnistgov/MIST #30: How to stitch microscope image without imagej software.

kuldeep203, 2022-12-02:
I am working on digital microscope. we don't have have commercial microscope , trying to make our own. i am stuck how to stitch images through code only. as of now i only have overlap raw image i dont have any other data. So let me know that i can stitch all images together to create a single image. Thanks for help in advance.

kvinicki, 2022-12-02:
You could write your own algorithm in python for example using FFT or
SIFT... or some deep learning solution like SuperGlue or LoFTR

tblattner, 2022-12-02:
MIST assumes a regular grid of overlapping images. It assumes a decently calibrated stage so that there isn't too much variability between image offsets. The algorithm we have in MIST does allow some wiggle room, and if there is sufficient content between the overlap regions in the images it also could have a better chance.

Have you tried to use MIST to stitch your images.

kuldeep203, 2022-12-05:
Thanks for the reply i haven't try as of now.i don't know where to start in mist because its tell to download imagej and i dont want any frontend tool i just need backend code to stitch images. i will pass images to function and it need to return stitch images.

kuldeep203, 2022-12-05:
> MIST assumes a regular grid of overlapping images. It assumes a decently calibrated stage so that there isn't too much variability between image offsets. The algorithm we have in MIST does allow [...]

can you tell me how i can use MIST in python.

tblattner, 2022-12-05:
MIST is currently only runnable through Fiji. It is possible to execute from command-line using a Fiji macro. See here for example of building a macro that calls MIST: https://github.com/usnistgov/MIST/wiki/Combined-User-Install-Guide#stitching-using-a-macro-script

We do not have a python implementation at this time, but that is something on our long-term lists. If you would like it sooner you can follow our algorithm here: https://github.com/USNISTGOV/MIST/wiki/assets/mist-algorithm-documentation.pdf

I would first test out stitching using MIST from within Fiji first to make sure the algorithm would work for you. You can take a look at the wiki for more information: https://github.com/usnistgov/MIST/wiki

closed 2023-01-25 by tblattner