usnistgov/MIST #31: After stitching i am getting output image in gray color not in original color.

kuldeep203, 2022-12-05:
i want output in same color as it was in input.
i have tried the cmd as shown:
ashlar "fileseries|/home/kuldeep/Downloads/Inferencing_with_Button/Image_Data/room|pattern=room{series:1}.jpg|overlap=0.2|width=10|height=1" --output-channels 0
but it is giving output in grey colour but i want it as the images was in the input.

tblattner, 2022-12-05:
What is the type of the image during input? At this time I believe we support Gray8, Gray16, Color_RGB, and Gray32. If none of those are found then it reverts back to FLOAT pixel type.

Can you share an example image grid?

kuldeep203, 2022-12-05:
thanks for the answer, please find the input images. tThese are the images taken from microscope.
[image]
[image]
[image]

tblattner, 2022-12-05:
Looking at the workflow, it does export it as color.  If you open the output image into Fiji it should look right.

You will need to enable SCIFIO to view it. In Fiji: Edit->Options->ImageJ2...  , then enable "Use SCIFIO when opening files (BETA!)".  You should also be able to open it in GIMP and do any conversions there as needed.

tblattner, 2022-12-05:
And just to be clear you are using MIST to stitch this right? I don't recognize your cmd.