usnistgov/MIST #33: Error to execute docker run with CUDA on MIST

Jeremias333, 2023-01-25:
I used the container that is in the MIST repository to run it. I was able to run the stitching for images with CPU (which is the --programType=java option), this time I want to run it with CUDA (using --programType=CUDA) as the parameter possibilities are documented in wipp-plugin.json and got this error:

![Exception in thread "Thread-1" java.lang.IllegalArgumentException: No enum constant gov.nist.isg.mist.lib.executor.StitchingExecutor.StitchingType.CUDA
        at java.lang.Enum.valueOf(Enum.java:238)
        at gov.nist.isg.mist.lib.executor.StitchingExecutor$StitchingType.valueOf(StitchingExecutor.java:94)
        at gov.nist.isg.mist.gui.params.utils.MacroUtils.loadMacroProgramType(MacroUtils.java:198)
        at gov.nist.isg.mist.gui.params.AdvancedParameters.loadMacro(AdvancedParameters.java:418)
        at gov.nist.isg.mist.gui.params.StitchingAppParams.loadMacro(StitchingAppParams.java:206)
        at gov.nist.isg.mist.lib.executor.StitchingExecutor.runStitchingWithMacro(StitchingExecutor.java:269)
        at gov.nist.isg.mist.lib.executor.StitchingExecutor.run(StitchingExecutor.java:193)
        at java.lang.Thread.run(Thread.java:748)](https://user-images.githubusercontent.com/38990510/214583653-666a08a0-6cdf-4b27-85c2-b4eefe2b5127.png)

The command I ran from docker was the one below:
`docker run  -v /opt/data:/data wipp/mist:2.0.7 --imageDir=/data/inputs2 --filenamePatternType=ROWCOL --filenamePattern="({r},{c}).jpg" --gridOrigin=UL --gridWidth=3 --gridHeight=7 --startTileRow=0 --startTileCol=0 --startRow=0 --startCol=0 --extentWidth=3 --extentHeight=7 --isTimeSlices=false --assembleNoOverlap=false --stageRepeatability=0 --overlapUncertainty=10 --programType=CUDA --outputPath=/data/outputs --numCPUThreads=6 --horizontalOverlap=50 --verticalOverlap=50 --outputFullImage=true`

At the level of curiosity is an image that I managed to assemble with CPU
[image]

tblattner, 2023-01-25:
Hello, due to modifications in GPU hardware the old GPU-accelerated stitching code is deprecated. In order to support modern hardware we would need to update the codebase. Unfortunately, we have not had the cycles to update the CUDA codebase. 

Here is the relevant code that we will need to update: https://github.com/usnistgov/MIST/blob/master/lib/jcuda/util-cuda-bin.cu

This issue goes into a little more detail: https://github.com/usnistgov/MIST/issues/25

We recommend using the FFTW version, which will have decent acceleration based on the number of CPU cores and CPU RAM that you have.

Sorry about that!

Jeremias333, 2023-01-25:
Is it the same for using CUDA 9? Because I'm using it

Jeremias333, 2023-01-25:
Gratitude, I will use it in FFTW for now.

closed 2023-01-25 by Jeremias333

Jeremias333, 2023-01-25:
I want to know what skills I need to have to contribute to this specific problem. I currently work with Python but I've worked with Java and also C.

tblattner, 2023-01-25:
The current functions are implemented in CUDA; however, there may be some high-level libraries to take advantage of such as CUB (https://github.com/NVIDIA/cub)

The functions that use implicit synchronization are the various reduce_max functions found in https://github.com/usnistgov/MIST/blob/master/lib/jcuda/util-cuda-bin.cu
Specifically: 
reduce_max_final
reduce_max_main
reduce_max_filter_final
reduce_max_filter_main

Originally these were used to find the N-max indices that are some distance from each other.  Later on we determined that the distance measurement wasn't really needed.

So what we'd need is a CUDA function (or library call) to find the N-max indices from the matrix.

The exact usage of these functions is found in:
https://github.com/usnistgov/MIST/blob/master/src/main/java/gov/nist/isg/mist/lib/imagetile/jcuda/CudaStitching.java

If you can help us with the CUDA kernel, I can work on creating the necessary bindings and update MIST to utilize it.

lucaslmrs, 2023-02-16:
Hello, we are trying to work on the reduce_max functions, how could we test it to verify if its ok to send you the new version?

Another thing:
I only know how to run the program by Docker or imageJ app. There is some way to run directly on java code?

tblattner, 2023-02-16:
For testing we do have the class: https://github.com/usnistgov/MIST/blob/master/src/test/java/gov/nist/isg/mist/jcuda/TestJCUDAMultiMax.java

Unfortunately it is not self-testing, and will need to be updated to pass the correct values. 
For example, two variables: file1 and file2, represent two neighboring tiles from a stitching experiment. Need to update the filepaths on these. There are a few other places in that code that will also need to be updated.

But could get that code to run to produce a phase correlation matrix (passed in to the multi peak function on line 141. If you save that matrix to file and then load it into your CUDA code, then that might be a way to check the results. Alternatively, you could modify that testing code to initialize a similar matrix between your code and Java, then verify they are getting identical results.

To develop MIST we use IntelliJ. IntelliJ is a convenient IDE that will easily understand Maven and install the necessary dependencies.  https://www.jetbrains.com/idea/

First you will want to make sure CUDA is setup on your machine. Here is an example of how to do that: https://github.com/usnistgov/MIST/wiki/Install-Guide#cuda-toolkit-v90-installation-on-windowslinuxmacosx.

I would use a later version of CUDA rather than version 9.0. It looks like the latest version supported by JCuda is 11.6.1, so I would target that version of CUDA (https://developer.nvidia.com/cuda-11-6-1-download-archive). http://javagl.de/jcuda.org/downloads/downloads.html

Then to load things into IntelliJ, clone the repo and open it in IntelliJ. This should get the code up and running. Then should be able to run the TestJCUDAMultiMax.java (after the necessary modifications).

As for creating a callable library from Java, I don't have published documentation on that. So once it comes to that step, I can assist to describe the process there or I would be happy to take the code and it do it from here.