usnistgov/MIST #34: Order of stitched tiles

kilir, 2023-02-27:
Hi,
during my imaging process, my samples become damaged which results in higher image intensities in the overlapping areas of individual tiles. 

The image acquisition goes snake/serpentine-like, such that if images are placed in a shingle-like way on top of each other, one does not see the damaged area.  Stitching with MIST works really great, however, I think I'm not able to control which tile would be placed on top (overlay in outpout options).

Question:  Is there any way I'd be able to control how images are placed (apart from overlay/linear/average)?

Cheers,
Rüdiger

mmajurski, 2023-02-27:
There are no controls for which images are laid over other images. 

Here is the code which controls the image generation.
[https://github.com/usnistgov/MIST/blob/master/src/main/java/gov/nist/isg/mist/lib/export/LargeImageExporter.java#L272](https://github.com/usnistgov/MIST/blob/master/src/main/java/gov/nist/isg/mist/lib/export/LargeImageExporter.java#L272)

It just iterates through the image grid laying things on top of each other (for overlay). 

You might want to explore using average blend, or a linear blend to soften and blend the edges together. 

The GUI elements to allow general control over which tiles are laid on top for overlay blend would be very complicated. 

If you are familiar with Python, using this 
[https://github.com/usnistgov/MIST/blob/master/assemble_stitched_image.py](https://github.com/usnistgov/MIST/blob/master/assemble_stitched_image.py)
and modifying it overlay like you want would probably be the easiest path forward.

tblattner, 2023-02-27:
One thing to add. The order in which the overlay is done is based on correlation. So the highest correlation tiles are laid last. The correlation is computed when we are computing relative displacements, and is a confidence. So the tiles that we have the most confidence are used last.

Another option would be to update MIST to pass in a priority per tile, then use the priorities when sorting the order. We do not have cycles to do that, but if you are interested I can point out the necessary places where this would need to be added.

Otherwise customizing the python code would be the best bet as pointed out by my colleague.

closed 2023-02-27 by kilir