usnistgov/MIST #36: Continuous stitching order incorrect

RyanEuesden, 2023-04-21:
Hello, I've recently noticed that the vertical continuous stitching algorithm has changed and does not stitch images in the correct orientation. I've checked an older version of MIST and this issue isn't present there so something has changed with an update.

[image]

Thanks,
Ryan

tblattner, 2023-04-21:
Hi Ryan,

Did the stitching fail when running this type of grid or is this a visualization issue in the GUI?

-Tim

RyanEuesden, 2023-04-21:
It also stitched the images from top to bottom.

Ryan

tblattner, 2023-04-21:
Hi Ryan,

Do you happen to have an example grid of images that I could take a look at?

-Tim

RyanEuesden, 2023-04-21:
I'll be able to produce some for you either later today or tomorrow.

Thanks!

tblattner, 2023-04-21:
Running our test suite it does appear to produce the right output, so this might just be a visualization bug in the GUI

Origin: Lower Right Direction: Vertical Continuous
F_013.tif F_012.tif F_005.tif F_004.tif 
F_014.tif F_011.tif F_006.tif F_003.tif 
F_015.tif F_010.tif F_007.tif F_002.tif 
F_016.tif F_009.tif F_008.tif F_001.tif 

Does this layout look right?

RyanEuesden, 2023-04-21:
That's the stitching order I expect. I'll triple check it's not my inputs.
Maybe the GUI bug has thrown me off.

Ryan

tblattner, 2023-04-21:
I've pushed an update that fixes the visualization. Feel free to ping here with additional information, and I can see if I can help with configuring the parameters and debugging some cases your working with.

RyanEuesden, 2023-04-22:
Yeah this was me being an idiot. My microscope was incorrectly set to vertical combing instead of snake.

At least the graphic has been fixed! This can be closed.

Ryan

tblattner, 2023-04-24:
No problem! And amazing that graphical glitch had been in there for years! So good catch on that! Glad its working well for you.

closed 2023-04-24 by tblattner