usnistgov/MIST #37: Issue with installation: no module named numpy

commandantcoustet, 2023-04-26:
Hello, 
I've been trying to install MIST for a couple of hours but "ImportError: no module named numpy" keep returning to me. I use the version 1.54d of Fiji on MacOS and installed both fftw and Cuda (but none of them is working). I read that similar issues are usually associated with the newer versions of Jython but I can't seem to find a solution to overcome this message. Does anyone have any idea how I could solve this issue?
Thank you very much in advance for your help!

cdtcoustet, 2023-04-28:
Thank you so much for your help! Here are the screenshots of the content of MIST folder (which I placed in the "plugin" repository) and the issue returned when I try to launch the plugin with FIJI: 
[image]
[image]
And I am not able to open the MIST interface.

tblattner, 2023-04-28:
Did you follow these instructions? https://github.com/USNISTGOV/MIST/wiki/Install-Guide

cdtcoustet, 2023-05-02:
Oops, you were right, I skipped one step and now everything works well... Thanks for your time!

closed 2023-05-02 by cdtcoustet