usnistgov/MIST #40: python branch

cwood1967, 2023-11-06:
What is the status of the python branch? I it ready to be used or tested.

Thanks,
Chris

mmajurski, 2023-12-08:
It is not complete/validated.

I will try to find some time before the new year to work on it and get that finished.

William859662, 2024-09-17:
I tried python version but I get   

```
File "C:\Users\public\Desktop\MIST-python\pciam.py", line 179, in compute_pciam
    y, x = np.unravel_index(ind, pcm.shape)
    ^^^^
ValueError: too many values to unpack (expected 2)
``` 
code is here 

```
        for ind in indices:
            y, x = np.unravel_index(ind, pcm.shape)
            if t1.r == t2.r:
                # same row, so compute NCC along Left-Right
                peak = PCIAM.peak_cross_correlation_lr(t1_img, t2_img, x, y)
            else:
                # different row, so compute NCC along Up-Down
                peak = PCIAM.peak_cross_correlation_ud(t1_img, t2_img, x, y)
            peak_list.append(peak)
```
Is it because of different Python version maybe ? I used 3.11.5 version of python

mmajurski, 2024-09-17:
I feel like its unlikely to be a python version problem. My first guess is that for your problem, you ended up with a pcm matrix which is 1D (i.e. either the width or height is 1).

Could you either share the data and the run parameters to enable debugging on my end, or you will need to debug on your end to figure out if the problem is the shape of the pcm matrix, at which point I would need to think about how to add checks against that edge case?

William859662, 2024-09-17:
Hello thank you,

I tried to modify this part ` y, x = np.unravel_index(ind, pcm.shape)` to ` y, x, _ = np.unravel_index(ind, pcm.shape)` and it did not give `ValueError: too many values to unpack (expected 2)` error

But this time it give `RuntimeError: No translations found in direction: VERTICAL`  in `File "C:\Users\public\Desktop\MIST-python\stage_model.py", line 70, in compute_overlap`

I tried to stitch with mist-java, and they were stitched

This is my cmdline:
`
 python .\main.py --image-dirpath C:\Users\public\Desktop\images --output-dirpath C:\Users\public\Desktop\images --grid-width 43 --grid-height 3 --filename-pattern "image{r}_{c}.jpg" --filename-pattern-type ROWCOL --grid-origin UL --numbering-pattern HORIZONTALCOMBING --horizontal-overlap 10 --vertical-overlap 10 --stage-repeatability 5 --time-slice 1
`

[This ](https://drive.google.com/drive/folders/1KcvfR0q5-6C3CPhW3vAjUPKAYFXAzkbw?usp=drive_link)is the images link

mmajurski, 2024-09-18:
You found a bug. I had forgotten to handle color images with the python version. The stitching only actual performs computation on greyscale images, so with any color inputs the Java side converts them to greyscale. The python side was missing that. 

I have added in the greyscale conversion, and the stitching works for me on your data. Please re-open the issue if you cannot get it to work.

closed 2024-09-18 by mmajurski