usnistgov/MIST #41: CUDA toolkit version

smith6jt-cop, 2023-11-11:
Hello,

Your instructions at https://github.com/USNISTGOV/MIST/wiki/Install-Guide indicate to use toolkit 9.0, but the help documentation in the FIJI plugin indicates toolkit 6.5.  Can you please clarify?

Thanks

smith6jt-cop, 2023-11-11:
Based on Windows version

closed 2023-11-11 by smith6jt-cop

tblattner, 2023-11-13:
Hello,

Unfortunately the CUDA version of MIST is no longer supported. There were issues related to synchronization introduced in newer versions of the GPU architecture that required us to rewrite some of the CUDA kernels. We have not had the cycles to add this capability. Thus far the multi-threading with FFTW has sufficiently kept up with disk I/O, and is scalable to the number of CPU cores on a machine.

smith6jt-cop, 2023-11-13:
Thank you for the information!  Yes the FFTW is working very well in our hands.

Thanks,

Justin Smith, PhD
Dept of Pathology, Immunology, and Laboratory Medicine
UF Diabetes Institute
College of Medicine
1275 Center Drive, 5th Floor
Gainesville, FL 32610

[External Email]