usnistgov/MIST #42: Multi-channel stitching

mbahin, 2023-12-05:
Hi,

Thanks for the nice tool! :)

I'm not sure I understand how I'm supposed to deal with multichannel images.

I have 40*52 microscope tiles with 4 channels each (BrightField and fluo).
It works well for BF channel. When I provide the whole dataset (BF is the first channel) it outputs a one-channel stitched image (BF).

I managed to re-use the positions obtained from BF stitching to apply them to another channel (I extracted this channel for all tiles) with the Multi-channel frame in the menu.
But is there a way to directly stitch the 4 channels at once, lead by what happens to the first channel? Is there something to do with the timeslices?

Cheers,
Mathieu Basin

tblattner, 2023-12-05:
Hi Mathieu,

First thanks for the compliments and the issue.

I think this would be an excellent feature to include. If I understand correctly, you have a folder of tiles that are being stitched, and each tile is stored with 4 channels? Do you happen to have example data available to play around with? 

I don't think I've encountered tiles with multi-channel like this yet. The mindset we had was each channel in this case would be a separate image rather than a single image. But I do think this storage methodology is completely valid. So I just need to take a look at how these tiles are represented when loading them.

Also, your methodology to reuse the stitching metadata to stitch the other channels is exactly the correct idea.

mbahin, 2023-12-06:
Also I have another problem (not related, maybe I should do another issue...).

I'm trying to run MIST as command-line on a server.
We use [Singularity](https://docs.sylabs.io/guides/3.5/user-guide/introduction.html) (Docker concurrent that can use Docker containers) so I pulled the container (`singularity pull MIST.sif docker://wipp/mist:latest`) which worked well.

Then tried to run a basic command-line:
`singularity run MIST.sif --imageDir=<path_to_BF_tiles>/ --filenamePatternType=SEQUENTIAL --filenamePattern=<filename_prefix>_{ppppp}.tif --gridOrigin=UL --gridWidth=6 --gridHeight=1 --outputPath=<path>/`

But I got the following error: `Error: Unable to access jarfile MIST.jar`
Should I install something first? Isn't is supposed to come with the container?

Cheers,
Mathieu

tblattner, 2023-12-06:
> Hi Timothy,
> 
> [Here](https://dl.espci.fr/ticket/84c1a9c39e16158b8f848483321c750d) you can download an example of 6 tiles that consist of 2 rows (3 tiles per row) of my dataset. The direction is `horizontal continuous` with the following order `30,450`, `30,451`, `30,452`, `30,511`, `30,512` and `30,513` (the 3 former being on top of the 3 latter). We have the BF as the first channel and then 3 fluo channels.
> 
> This is not so complicated on my side to build a workflow to process them as I want (slicing my tiles into each channel -> determining the stitching parameters from BF -> applying the same stitching to other channels -> merging them back) but I hope this could be integrated to your tool easily.
> 
> Cheers, Mathieu

Thanks for the example dataset. I've downloaded it and will take a closer look at how we view this dataset. My hope is that I can expose the different channels into MIST so you can select which channel to stitch and then automatically apply that stitching vector to the remainder of the image, merging it into a new stitched TIFF . I don't have a good timeline for when I will have the cycles to work on it though. Currently I'm a bit oversubscribed. In the meantime, I would definitely utilize your current workflow, as I would not wait for the new feature to make it into MIST.

MyleneSimon, 2023-12-06:
> It was actually better with 2.0.9 version!

Thank you for testing!

> It seems that the `--gridDirection` (found [here](--gridDirection)) doesn't exist in this version but I have trouble to find the equivalent.

I think that the grid direction param is actually called `numberingPattern` in the MIST cli, but I see `--gridDirection` mentioned in the wiki indeed, we may have to update that section of the documentation. 
Also, the format for passing the arguments to the CLI should be `--argName argValue` as opposed `--argName=argValue`, in case that gives you issues as well.

MyleneSimon, 2023-12-07:
Hi Mathieu,

> Hi Mylene,
> 
> I replaced by `numberingPattern` which solves the problem but I still don't get the expected output. It seems that it can't find my images: `Unable to find any images in the image grid.` and I also notice that it says that it does a sub grid (`MIST is stitching a Sub Grid.`), maybe because I'm getting wrong in the parameters (because this is not what I want to do if this is the same as te the `subgrid` tab on FIJI GUI version). I guess I have a problem in the way I indicate it the way to locate the images (I double checked the paths). I'm not sure whether it's important to use quotes around some of the arguments (there are around the `filenamePattern` in the `docker_example.sh` file, which would make sense since it's a sort of regex).
> 
> My current command line is: `singularity run MIST.sif --imageDir <path_to_BF_tiles>/ --filenamePatternType SEQUENTIAL --filenamePattern "<filename_prefix>_{ppppp}.tif" --gridOrigin UL --numberingPattern HORIZONTALCONTINUOUS --startTile 30450 --gridWidth 6 --gridHeight 1 --outputPath <path>/`
> 

I am thinking a few possibilities here as to why it is not finding the images:
- is the data folder mounted in the container? See [Singularity doc on binds](https://docs.sylabs.io/guides/3.0/user-guide/bind_paths_and_mounts.html#specifying-bind-paths). Depending on the Singularity configuration, $HOME might be mounted by default, but that is not always the case. So you could try adding a bind (`--bind </path-to-data>:</mount-path-in-container>`) and then make sure that the `--imageDir` and `--outputPath` paths are matching the bind/mount path (so for example if the data is located at `/home/user/folderA/myImages` and your bind is `--bind /home/user/folderA:/data`, then you would specify `--imageDir /data/myImages` and `--outputPath /data/output`). 
- The quotes should not be needed for the filename pattern
- I am not sure why you get the message about the sub-grid (@tblattner or @mmajurski probably have more insight on that), but I checked on our automated workflows and I see that we set the `extentWidth` and `extentHeight` to match the grid width and height by default.

> Also I realised that I'm doing it on the server (so without monitor) and I saw that there is an argument `headless` in the help (for which I'm not sure what value I should set). Maybe I should use it? Although my problem is before that I think.
> 
> Cheers, Mathieu

The Docker image is already configured to run headless.

tblattner, 2023-12-07:
I think there might be a bug with extentWidth and extentHeight. When we load parameters from command-line it doesn't appear to match up with how we load the parameters from the GUI. So for now I would suggest specifying extent-width and extent-height options to match your grid-width and grid-height, because it looks like the default is to set them each 1.

I have a conflict throughout today, but will try to push out a fix later tonight. This would require the version numbering to be incremented and docker hub updated though.

mbahin, 2023-12-07:
Hi both,

Thanks for the high reactivity, very appreciated! :)

@MyleneSimon, you had it with your first guess, this was a rookie Singularity mistake! I bound appropriately and it found the images fine!

Then I had other issues:

```
OpenJDK 64-Bit Server VM warning: You have loaded library /tmp/BridJExtractedLibraries6846359835200471305/libbridj.so which might have disabled stack guard. The VM will try to fix the stack guard now.
It's highly recommended that you fix the library with 'execstack -c <libfile>', or link it with '-z noexecstack'.
```
=> maybe this is something on the configuration of the server I use. Not sure whether I really have to deal with it to make it work.

```
Warning: no good translations found for North direction. Estimated translations generated from the overlap.
Warning: no good translations found for North direction. Repeatability has been set to zero.
```
=> and same for West, although it worked well through the GUI on FIJI, and moreover, the announced computed overlap is not stable if I run the same command several times.

No stitched image is produced (I guess because it didn't found "good translation"). I attach [img-statistics.txt](https://github.com/usnistgov/MIST/files/13602519/img-statistics.txt) in case it can help (and by the way my test is on the 6 first tiles of the dataset, first row, of which you have the 3 first in what I sent if you want to test).

@tblattner ok for the headless mode and I tried with `--extentWidth 6 --extentHeight 1`, it made the message disappear.

Cheers,
Mathieu

mbahin, 2023-12-08:
Hi,

It actually worked! I wasn't aware that the result wasn't saved. I added `--outputFullImage True` to the command-line.

About the North/West overlap, I thought it was the overlap, side and top/bottom, between the tiles that is determined to create a good stitching but maybe I was wrong. Or this is not something stable although the result is still nice.

It's amazing, I was able to get a nice stitch for 32x10 tiles in 20 seconds, great work!

My dataset is actually 32x25 tiles, each tile is 2304x2304 (~10MB), the memory crashed if I try with more than 10 rows from what I tried with the following message when writing the full image:
```
Error: Insufficient memory to save image.
Out of memory error: Java heap space
```
Is it possible for the user to enlarge the JVM memory in some way?

I also have an acquisition of the same device in 4x (above is related to 20x magnification tiles) that I'm also interested to stitch. It's 7x5 tiles, ~5MB each.
I tried to stitch the 4x dataset. It works well for the first row alone with the following command:
`singularity run --bind <local_path>:<remote_path> MIST.sif --imageDir <path_to_BF_tiles>/4x/ --filenamePatternType SEQUENTIAL --filenamePattern "<filename_prefix>_{ppppp}.BF.tif" --gridOrigin UL --numberingPattern HORIZONTALCONTINUOUS --startTile 30205 --gridWidth 7 --extentWidth 7 --gridHeight 1 --extentHeight 1 --blendingMode LINEAR --blendingAlpha 3 --outputPath <path>/ --outputFullImage True --outFilePrefix 164_4x_BF_ --headless True`
But it failed for the whole dataset (and actually as soon as I tried with more than 1 row, changing `--gridHeight` and `--extentHeight` parameters) with the following problem:
```
Warning: the computed repeatability (54) is unusually large. Consider manually specifying the repeatability in the Advanced Parameters.
Repeatability for West: 54 pixels
Calculated Repeatability: 109 pixels
```
I tried with some values 0 (appear not to be considered), 3, 5 but it didn't change the output. I get a an image with the appropriate number of rows but columns seem to have a huge overlap (making ~3 instead of 7). I understood that this is the uncertainty of the stage move so it should be a low value (like it is when it works well). [Here](https://dl.espci.fr/ticket/db72451701f4292a6a000845f7fa47fe) is a link to download the dataset if you want to check.

A final feedback that might be interesting for you: I had to specify `--headless True` to find `headless: true` in the statistics file (seems to be `False` by default, at least in the file, it wasn't a actual problem).

Sorry to ask heavy support! :)

Cheers,
Mathieu

mbahin, 2023-12-14:
Hi @MyleneSimon,

I'm currently exploring this problem. I'm not sure yet but I think that Singularity limits the memory (the option is more to constraint memory to lesser than what the host system can do). I'll dig further and let you know.

Cheers,
Mathieu