usnistgov/MIST #46: Parallelization Linear Blending

William859662, 2024-09-17:
Hi,

Is it possible to use all core when doing linear blending ? Because Linear blend is taking much time, If I use Overlay blend, stitch result is not good as linear blending result like below images ; my overlap ratio is %10 i tried with %15 but still same result

How to improve overlay blend result or accelerate linear blending ?

Please see below images, they stitched with same parameter only blending is overlay and linear, as you see result is different, how can I correct this ?

[image]
[image]

tblattner, 2024-09-17:
It might be possible to parallelize the blending functionality. Looking at this loop, https://github.com/usnistgov/MIST/blob/878709a00f0a31f75d2a827cb3274f2d26a9afbb/src/main/java/gov/nist/isg/mist/lib/export/LargeImageExporter.java#L268

If we were to distribute these loops to threads then the blend call could run in parallel, and then sequentially apply post processing. I don't have the cycles to work on this optimization right now though, but if you have the ability, I'd happy to help guide an implementation and review any PRs.

William859662, 2024-09-17:
Thank you,

Although I am not familiar with java, I think I can do this, I will look into it at a suitable time. 

As far as I understand, we can only parallelize “Blend Call Time”, right? For example, “Init Time” does not change ?

Well, I think overlay and linear blending affect the stitch result a bit, can you please tell me the reason for this? For example, is the difference in the picture I shared related to blending or is it a different situation ?

tblattner, 2024-09-17:
The image you showed is a result of the blending only. The stitching is the same between both. So its possible the last tile selected during overlay was not the optimal causing this seam. The linear blend will smooth it across all overlapping tiles.

The way I would try to implement the parallelization would be to have one TileBlender per thread, then gather all relevant images to build that tile. So the init call should only happen once, but a reset might be necessary to reuse the memory within the TileBlender for the next tile.

The blend call should be able to be done asynchronously across all threads.

The postprocess step requires writing to the omeTiff object, which I am not sure if that is thread safe or not. So most likely the postprocess would need to be done sequential.

Thanks for taking a look at this!

William859662, 2024-10-19:
Thank you very much @tblattner, frankly I will also examine the code and I will try to parallelize it, maybe I can even use gpt because I don't have much java knowledge but I will examine code and try to increase speed of blending because Linear blending gives really good results, I couldn't get such good results in other blend methods. stitch process is fast on a good cpu but as you know linear blend takes more time but give very good result