usnistgov/MIST #47: Installation on Windows for CLI

mbahin, 2025-08-15:
Hi,

I already use your (nice !) tool on Unix (from command-line) but, but I would like to use it on a Windows environment from command-line.
Is this possible? I don't see this on the documentation.

Cheers,
Mathieu

mbahin, 2025-08-15:
Hi Tim,

I actually never really installed it, even for Unix env, I pulled the Singularity container (which works perfectly for me in this environment).

From what I understand [here](https://github.com/usnistgov/MIST/wiki/Combined-User-Install-Guide#install-guide), I would first have to find a way to install FFTW (following something like [that](https://www.fftw.org/install/windows.html)) and CUDA (following [here](https://github.com/usnistgov/MIST/wiki/Combined-User-Install-Guide#cuda-toolkit-v65-installation-on-windowslinuxmacosx)).
Then it's not very clear to me what I should do. Download the repo and use a file (which one?) from it in some way with command-line?

Cheers,
Mathieu

tblattner, 2025-08-15:
Another option that we discussed over email was to use the python port of MIST: https://github.com/usnistgov/MIST/tree/python

closed 2025-08-15 by tblattner